barracuda-fsh/pyobd #1: Auto diagnostics? You mean car diagnostics?

mitar, 2023-09-14:
I had hard time understanding what this package is about. "auto diagnostics" to means something else (automatic diagnostic .. of what). You probably want to say "car diagnostics"?

patgd, 2023-09-15:
Auto contracts automobile.

robheffo79, 2023-09-17:
Coming from Australia, I can see the confusion. To me also "Auto Diagnostics" reads Automatic Diagnostics.
We don't call vehicles "Automobiles", we call them "Cars".

barracuda-fsh, 2023-09-17:
I didn't know. Thank you. I'll use cars then.

AtesComp, 2023-09-17:
Errr, vehicles?

patgd, 2023-09-17:
[Auto: usually four-wheeled [automotive](https://www.merriam-webster.com/dictionary/automotive) vehicle designed for passenger transportation](https://www.merriam-webster.com/dictionary/auto).

robheffo79, 2023-09-19:
> > [_Auto_: a four-wheeled vehicle designed for passenger transportation](https://www.merriam-webster.com/dictionary/auto).

While technically true, automobile is pretty much only used in the United States. Even the Wikipedia entry for Automobile redirects to Car. So, the audience who won't be confused at least initially by the name will be dwarfed by the audience who will.

closed 2023-09-20 by barracuda-fsh