barracuda-fsh/pyobd #2: Where to find the standalone executables ?

Chralu, 2023-09-15:
Hello,

thanks for that promising project !

The Readme indicates there are standalone executables to download, but I can't find them :/
Could you give more indications on where to find them ?

In the meantime, I'll try running the script

Thank you :)

patgd, 2023-09-15:
Read through the rest of the README.md. There is a section on [Creating the Executable](https://github.com/barracuda-fsh/pyobd#creating-the-executable).

barracuda-fsh, 2023-09-15:
I used to have tthem under downloads, but I will have to make new ones. The Linux version is there, I have to make Mac and Windows version. Coming ASAP.

nevi-me, 2023-09-15:
> I used to have tthem under downloads, but I will have to make new ones. The Linux version is there, I have to make Mac and Windows version. Coming ASAP.

Thank you, also only saw the Linux one

barracuda-fsh, 2023-09-16:
Now there's a new version 1.15 and all the executables with it. I would be glad if you try it out.

closed 2023-09-20 by barracuda-fsh