barracuda-fsh/pyobd #5: Please consider making pyobd PEP517 compliant by adding pyproject.toml, etc, and placing it on https://pypi.org ?

yurivict, 2023-09-16:
This would make it easy to create ports/packages for it, and this would make it easier to install for users.

Thanks,
Yuri

Kydlaw, 2023-09-17:
Maybe barracuda-fsh doesn't want to create a lib out of this code.

But that would be interesting to define at least the minimum version of Python required in this project.

yurivict, 2023-09-17:
> Maybe barracuda-fsh doesn't want to create a lib out of this code.

It's not about making it into a lib.
PEP517 allows to install executables in a uniform way which makes it easy to install using standard methods.

gray-adeyi, 2023-09-26:
The name pyobd and obd is already taken on pypi

yurivict, 2023-09-26:
@gray-adeyi 

Just name it pyobd2 or obd2 then.
Many other projects solve this problem this way.

Kydlaw, 2023-09-26:
I updated the PR to use pyobd2

PR #7 (2024-06-08): Add pyproject.toml
Fixes #5

Bare pyproject.toml file.
As there is no build system attached to the current repository, it uses the [build](https://pypi.org/project/build/) system recommended in the setuptools documentation.

The minimal python version is debatable.
commit 285705e5 (2024-06-08): Please consider making pyobd PEP517 compliant by adding pyproject.toml, etc, and placing it on https://pypi.org ?
Fixes #5
closed 2024-06-08 by barracuda-fsh