barracuda-fsh/pyobd #13: issues on rasberry pi os

koatic, 2023-12-15:
first thanks for rebooting the pyobd project!!

Trying to get it to run on raspberry pi with arm processor so can't use your EXE's.
First had issues with numpy
"ImportError: Unable to import required dependencies: numpy
Original error was: libcblas.so.3: cannot open shared object file: No such file or directory"

Fixed numpy dependancy with "sudo apt-get install libatlas-base-dev" That package seems to include the missing libcblas.so.3

now running script, I get the following output.
user@raspberrypi:~/pyobd $ python3 pyobd.py
Traceback (most recent call last):
  File "/home/user/pyobd/pyobd.py", line 53, in <module>
    import obd_io  # OBD2 funcs
  File "/home/user/pyobd/obd_io.py", line 39, in <module>
    import obd
  File "/home/user/pyobd/obd/__init__.py", line 41, in <module>
    from .obd import OBD
  File "/home/user/pyobd/obd/obd.py", line 38, in <module>
    from .commands import commands
  File "/home/user/pyobd/obd/commands.py", line 36, in <module>
    from .decoders import *
  File "/home/user/pyobd/obd/decoders.py", line 38, in <module>
    from .UnitsAndScaling import Unit, UAS_IDS
  File "/home/user/pyobd/obd/UnitsAndScaling.py", line 33, in <module>
    import pint
  File "/home/user/.local/lib/python3.9/site-packages/pint/__init__.py", line 28, in <module>
    from .formatting import formatter, register_unit_format
  File "/home/user/.local/lib/python3.9/site-packages/pint/formatting.py", line 20, in <module>
    from .babel_names import _babel_lengths, _babel_units
  File "/home/user/.local/lib/python3.9/site-packages/pint/babel_names.py", line 11, in <module>
    from .compat import HAS_BABEL
  File "/home/user/.local/lib/python3.9/site-packages/pint/compat.py", line 41, in <module>
    from typing_extensions import Self  # noqa
ImportError: cannot import name 'Self' from 'typing_extensions' (/usr/lib/python3/dist-packages/typing_extensions.py)

I am novice on the raspberry so I am not sure it I am doing something wrong or it just wont run on the rasberry pi OS debian version.
Thanks for your work.

barracuda-fsh, 2023-12-15:
I have no idea right now what seems to be the problem. 

Does this work?
pip install -r requirements.txt

koatic, 2023-12-15:
This issue is caused by an outdated version of the typing-extensions module.
I was able to fix this and get it running on the Raspberry Pi using "pip install typing-extensions --upgrade"
Thanks again for your work on this project!

closed 2023-12-15 by koatic

barracuda-fsh, 2023-12-16:
I'm glad you found the solution. And it's nice to be appreciated. Thanks.