barracuda-fsh/pyobd #15: [Help needed] I can't connect the OBD-II

louisonsarlinmagnus, 2024-01-29:
Hi,

I'm using a [USB OBD-II Connector](https://www.amazon.fr/dp/B08ZDB5JBS?psc=1&ref=ppx_yo2ov_dt_b_product_details), Ubuntu 22.04 and I can't get it to work with your software (I'm using the executable in version 1.15 release on GitHub).

I've manage to connect using Putty with these settings : 
[image]

But when I try with PyOBD using the same settings, it won't connect : 
[image]

Do you have an idea of what I'm doing wrong ?

Thank you !

barracuda-fsh, 2024-01-29:
Maybe increase timeout and set auto baud rate?

barracuda-fsh, 2024-01-29:
And maybe flip the switch if it has one.

louisonsarlinmagnus, 2024-01-29:
I set everything to auto and flipping the switch and still can't connect. Is there any way to get some logs ?

barracuda-fsh, 2024-01-29:
Run it from terminal or command prompt and you should get output. You can paste it here.

louisonsarlinmagnus, 2024-01-29:
Nice !!
I run it with sudo in a terminal and worked like a charm !!! 
Thanks a lot !!

Do you think it could be possible to give more information on the GUI when connection failed ?

closed 2024-01-29 by louisonsarlinmagnus