barracuda-fsh/pyobd #16: Detected: Trojan:Win32/Wacatac.B!ml

thedapperissue, 2024-02-20:
[image]

barracuda-fsh, 2024-02-21:
Yes, it's a false positive. It's made solely from the code you see here.

barracuda-fsh, 2024-02-23:
I rebuilt the executables in console mode now. Tell me if it still happens. I think it's OK now.

barracuda-fsh, 2024-04-18:
Maybe someone can find a solution instead of pointing to what we already know? It's a false positive.

pxFIN, 2024-04-18:
> Maybe someone can find a solution instead of pointing to what we already know?

Sorry barracuda i was trying to answer your question about 'tell me if it still happens' not trying to be negative

I remember years back i was using pyinstaller and yes this same challenge was a problem.. I cant think of a easy solution for it

I think its quite well explained on this post

https://discuss.python.org/t/pyinstaller-false-positive/43171/11

barracuda-fsh, 2024-04-18:
Ok, this sounds different. Thank you, I'll read that post.

barracuda-fsh, 2024-04-18:
I looked into it and rebuilt the Windows executable. I think now it's reported it's good. Please correct me if I'm wrong. Please do, because I want to know for sure.

pxFIN, 2024-04-18:
> I looked into it and rebuilt the Windows executable. I think now it's reported it's good. Please correct me if I'm wrong. Please do, because I want to know for sure. Thanks to everyone for letting me know.

Downloaded zip file, extracted files.. windows complains once but when you 'run anyway' app starts ok.. when you start app again no warnings from windows..  zero warnings from f-secure internet security

closed 2024-04-20 by barracuda-fsh