barracuda-fsh/pyobd #17: meet requirements on gentoo

darkicekeel, 2024-03-06:
hello there, i was trying to find every package required on gentoo, can't really find "pint"
may you provide more details, so i can find the right package?

moreover, do you recommend to use conda to manage pyobd?

darkicekeel, 2024-03-06:
i'm trying right now with a python virtual env and i managed to install pint from there, i'll see if this works

darkicekeel, 2024-03-14:
ok it keeps failing while compiling wxpython 4.1.1

` [  8/921] Compiling sip/siplib/descriptors.c                                                                                                                                                                     
  ../../../../sip/siplib/siplib.c: In function 'sip_api_get_frame':                                                                                                                                                
  ../../../../sip/siplib/siplib.c:13750:22: error: invalid use of undefined type 'struct _frame'                                                                                                                   
  13750 |         frame = frame->f_back;                                                                                                                                                                           
        |                      ^~                                                                                                                                                                                  
                                                                                                                                                                                                                   
  Waf: Leaving directory `/tmp/pip-install-_6saj4ui/wxpython_ea8b7700753941f6acd397fb73303d80/build/waf/3.1/gtk3'                                                                                                  
  Build failed                                                                                                                                                                                                        -> task in 'siplib' failed with exit status 1 (run with -v to display more information)                                                                                                                         
  Command '"/home/user/software/pythonvirtualenv/bin/python" /tmp/pip-install-_6saj4ui/wxpython_ea8b7700753941f6acd397fb73303d80/bin/waf-2.0.19 --wx_config=/tmp/pip-install-_6saj4ui/wxpython_ea8b7700753941f6
acd397fb73303d80/build/wxbld/gtk3/wx-config --gtk3 --python="/home/user/software/pythonvirtualenv/bin/python" --out=build/waf/3.1/gtk3 configure build ' failed with exit code 1.                              
  Finished command: build_py (0m4.983s)                                                                                                                                                                            
  Finished command: build (25m24.631s)                                                                                                                                                                             
  Command '"/home/user/software/pythonvirtualenv/bin/python" -u build.py build' failed with exit code 1.                                                                                                       
  [end of output]                                                                                                                                                                                                  
                                                                                                                                                                                                                   
  note: This error originates from a subprocess, and is likely not a problem with pip.                                                                                                                             
  ERROR: Failed building wheel for wxpython                                                                                                                                                                        
  Running setup.py clean for wxpython                                                                                                                                                                              
Failed to build wxpython                                                                                                                                                                                           
ERROR: Could not build wheels for wxpython, which is required to install pyproject.toml-based projects   `

barracuda-fsh, 2024-03-15:
You're probably missing something. wxpython often needs some libraries to compile it. I don't know how it is on gentoo. Maybe you google and find a solution and post it here? I don't have gentoo so it will be hard for me to do it. And post the solution here, please. Thanks.

darkicekeel, 2024-03-16:
already searched, can't find the missing point.
i'll update if i find something
btw, i had to use a local virtual env for pip like explained here https://wiki.gentoo.org/wiki/Pip
`To install package using the pip command, first create and activate a virtual environment:

user $python -m venv /path/to/venv
user $. /path/to/venv/bin/activate
Then install into the virtual environment:

user $pip install <package>
To exit the virtual environment, run:

user $deactivate
The virtual environment is not deleted, and can be re-entered by re-sourcing the activate file (with the command beginning with the period).`

with this, some of requirements.txt went ok but it stopped on that particular wxpython

barracuda-fsh, 2024-03-16:
Do you have all these libraries installed?

sudo apt-get install dpkg-dev build-essential libjpeg-dev libtiff-dev libsdl1.2-dev libgstreamer-plugins-base1.0 libnotify-dev freeglut3 freeglut3-dev libsm-dev libgtk-3-dev libwebkit2gtk-4.0-dev libxtst-dev gettext python3-dev python3-pip
sudo apt install libglib2.0-dev libsmbclient-dev libcups2-dev libgirepository1.0-dev libcurl4-openssl-dev libssl-dev libsystemd-dev librsync-dev

darkicekeel, 2024-03-19:
trying to figure out all the corresponding packages on gentoo, but it seems most of all is there

[I] app-arch/dpkg
     Available versions:  1.20.9-r1^t ~1.20.12-r1^t ~1.21.1^t ~1.21.15-r2^t ~1.21.15-r3^t **1.21.19^t **1.21.19-r1^t {+bzip2 libmd +lzma nls selinux static-libs test unicode +update-alternatives +zlib +zstd}
     Installed versions:  1.20.9-r1^t(05:56:12 19/03/2024)(bzip2 lzma nls unicode update-alternatives zlib -libmd -selinux -static-libs -test)
     Homepage:            https://packages.qa.debian.org/dpkg
     Description:         Package maintenance system for Debian

[I] media-libs/libjpeg-turbo
     Available versions:  2.1.5.1(0/0.2) 3.0.0(0/0.2) ~3.0.1(0/0.2) ~3.0.2(0/0.2) {java static-libs ABI_MIPS="n32 n64 o32" ABI_S390="32 64" ABI_X86="32 64 x32" CPU_FLAGS_ARM="neon"}
     Installed versions:  3.0.0(0/0.2)(02:27:25 17/03/2024)(-java -static-libs ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="32 64 -x32" CPU_FLAGS_ARM="-neon")
     Homepage:            https://libjpeg-turbo.org/ https://github.com/libjpeg-turbo/libjpeg-turbo
     Description:         MMX, SSE, and SSE2 SIMD accelerated JPEG library
     
[I] media-libs/tiff
     Available versions:  4.5.0-r2(0/6)^t 4.5.1(0/6)^t 4.6.0(0/6)^t {+cxx jbig jpeg lzma static-libs test verify-sig webp zlib zstd ABI_MIPS="n32 n64 o32" ABI_S390="32 64" ABI_X86="32 64 x32"}
     Installed versions:  4.6.0(0/6)^t(02:37:35 17/03/2024)(cxx jpeg lzma zlib -jbig -static-libs -test -verify-sig -webp -zstd ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="32 64 -x32")
     Homepage:            http://libtiff.maptools.org
     Description:         Tag Image File Format (TIFF) library

[I] media-libs/libsdl2                                                                                   
     Available versions:  2.28.1 2.28.3 ~2.28.4-r1 2.28.5^t ~2.28.5-r1^t {X alsa aqua custom-cflags dbus doc fcitx4 gles1 gles2 (+)haptic ibus jack +joystick kms libsamplerate nas opengl oss pipewire pulseaudio 
sndio +sound static-libs test +threads udev +video vulkan wayland xscreensaver ABI_MIPS="n32 n64 o32" ABI_S390="32 64" ABI_X86="32 64 x32" CPU_FLAGS_PPC="altivec" CPU_FLAGS_X86="3dnow mmx sse sse2" VIDEO_CARDS="
vc4"}                                                                                                    
     Installed versions:  2.28.5^t(12:36:06 01/03/2024)(X alsa dbus haptic joystick nas opengl pulseaudio sound threads udev video vulkan -aqua -custom-cflags -doc -fcitx4 -gles1 -gles2 -ibus -jack -kms -libsamp
lerate -oss -pipewire -sndio -static-libs -test -wayland -xscreensaver ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="32 64 -x32" CPU_FLAGS_PPC="-altivec" CPU_FLAGS_X86="mmx sse sse2 -3dnow" VIDEO_CARDS="
-vc4")                                                                                                   
     Homepage:            https://www.libsdl.org/                                                                                                                                                                  
     Description:         Simple Direct Media Layer 

[I] media-libs/libsdl                                                                                    
     Available versions:  1.2.15_p20221201 1.2.64^t 1.2.64_p20230726^t 1.2.68^t {X aalib alsa custom-cflags dga fbcon +joystick libcaca nas opengl oss pulseaudio +sound static-libs test tslib +video xinerama xv 
ABI_MIPS="n32 n64 o32" ABI_S390="32 64" ABI_X86="32 64 x32"}                                                                                                                                                       
     Installed versions:  1.2.68^t(02:28:18 17/03/2024)(X alsa joystick opengl sound video -test ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="32 64 -x32")                                                
     Homepage:            https://github.com/libsdl-org/sdl12-compat                                                                                                                                               
     Description:         Simple Direct Media Layer 1.2 compatibility wrapper around SDL2  

[I] media-libs/gstreamer
     Available versions:  (1.0) 1.20.5^t 1.20.6^t ~1.22.3^t
       {+caps +introspection nls +orc test unwind ABI_MIPS="n32 n64 o32" ABI_S390="32 64" ABI_X86="32 64 x32"}
     Installed versions:  1.20.6(1.0)^t(21:56:42 27/02/2024)(caps introspection nls orc -test -unwind ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="32 64 -x32")
     Homepage:            https://gstreamer.freedesktop.org/
     Description:         Open source multimedia framework

[I] x11-libs/libnotify
     Available versions:  0.8.3^t {gtk-doc +introspection test ABI_MIPS="n32 n64 o32" ABI_S390="32 64" ABI_X86="32 64 x32"}
     Installed versions:  0.8.3^t(02:39:47 17/03/2024)(introspection -gtk-doc -test ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="32 64 -x32")
     Homepage:            https://gitlab.gnome.org/GNOME/libnotify
     Description:         A library for sending desktop notifications

[I] media-libs/freeglut
     Available versions:  3.4.0 {ABI_MIPS="n32 n64 o32" ABI_S390="32 64" ABI_X86="32 64 x32"}
     Installed versions:  3.4.0(02:38:23 17/03/2024)(ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="32 64 -x32")
     Homepage:            https://freeglut.sourceforge.net/
     Description:         A free OpenGL utility toolkit, the open-sourced alternative to the GLUT library

[... cut ...]
     Available versions:  3.2.4-r3 3.2.7-r2 3.2.7-r3 3.2.7-r4 **9999*l {acl examples iconv ipv6 lz4 rrsync ssl stunnel system-zlib verify-sig xattr xxhash zstd PYTHON_SINGLE_TARGET="(+)python3_10 python3_11"}
     Installed versions:  3.2.7-r4(06:31:34 06/03/2024)(acl iconv ssl xattr -examples -lz4 -rrsync -stunnel -system-zlib -verify-sig -xxhash -zstd PYTHON_SINGLE_TARGET="python3_10 -python3_11")
     Homepage:            https://rsync.samba.org/
     Description:         File transfer program to keep remote files into sync

-------------------------

these the corresponding packages i'm finding. may be prone to some mistake or mismatch
hopefully this will help anyway

istride, 2024-03-28:
I had a similar issue, but on Arch Linux. What worked for me, in the end, was:

- using Python 3.11
- remove all the version restrictions in requirements.txt - install more recent versions of packages

I went on a little detour using different versions of Python, using pyenv, and managed to get things to work on Python 3.10. Some extra care is necessary when using pyenv; maybe these things are also relevant with Gentoo.

Python must be installed with the "--enable-shared" option.
```
env PYTHON_CONFIGURE_OPTS="--enable-shared" pyenv install -f 3.10.13
```

It's not enough just to set the local Python. Must also, or instead, set the global Python when building wxPython.
```
pyenv local 3.10.13
pyenv global 3.10.13
```

Hope this helps.

darkicekeel, 2024-04-16:
i did a last try 2 days ago, updating all requirements to the latest version and wxpython this time was correctly compiled! maybe they fixed something ... good news!
then i compiled the program and it went ok too.
i was able to see the gui, next i'll try and see if the obd usb device will work when plugged into the serial port.
so far, it's ok right now as it is.

of course, i worked all the time with the python env activator, just to remind it to the readers

barracuda-fsh, 2024-04-16:
That's good news. It looks like I can close the issue. Thanks.

closed 2024-04-16 by barracuda-fsh