barracuda-fsh/pyobd #18: Error in byte to DTC conversion

yazyr, 2024-05-06:
There seems to be some logic error in converting byte values to dtc. The logic is present in the 'decoders.py' script under the obd folder. I don't know what is the logic in converting byte values to dtc codes. If anyone has any idea, could you please guide me? It would be a huge favor.

barracuda-fsh, 2024-05-06:
Is it possible that this is an error in the obdsim? I don't know.

barracuda-fsh, 2024-05-06:
If someone finds a solution I will be greatful to you both.

yazyr, 2024-05-06:
Same error was observed in Skoda Octavia Mk-II. Unplugging the MAP sensor coupler provided me an irrelevant error code. Also the code snippet that carries out the byte to dtc conversion has some comments that explain how the conversion is being done, which actually doesn't make sense. The binary equivalent of hexadecimal examples they have picked looks wrong. Or I am not sure I am interpreting the comments of the author properly. I have attached the code snippet performing the conversion below:

def parse_dtc(_bytes):
    """ converts 2 bytes into a DTC code """

    # check validity (also ignores padding that the ELM returns)
    if (len(_bytes) != 2) or (_bytes == (0, 0)):
        return None

    # BYTES: (16,      35      )
    # HEX:    4   1    2   3
    # BIN:    01000001 00100011
    #         [][][  in hex   ]
    #         | / /
    # DTC:    C0123

    dtc = ['P', 'C', 'B', 'U'][_bytes[0] >> 6]  # the last 2 bits of the first byte
    dtc += str((_bytes[0] >> 4) & 0b0011)  # the next pair of 2 bits. Mask off the bits we read above
    dtc += bytes_to_hex(_bytes)[1:4]

    # pull a description if we have one
    return (dtc, DTC.get(dtc, ""))
    
The above code snippet could be found in the decoders.py file inside the 'obd' folder

barracuda-fsh, 2024-05-25:
I used the code from older pyobd for dtc parsing. Can you test if it works properly now? I don't know how to trigger a dtc code in simulator.

barracuda-fsh, 2024-05-25:
Please do it as soon as possible. You said it would be a huge favor to fix is, so let's do it.

barracuda-fsh, 2024-05-26:
I can test it myself now with obdsim. It's not working properly yet, but I will fix it.

barracuda-fsh, 2024-05-28:
I can confirm that def parse_dtc(_bytes): function is working properly. The bug is somewhere else.

barracuda-fsh, 2024-05-28:
https://github.com/brendan-w/python-OBD/issues/61

I changed decoders.py

d += message.data[2:]

to this

d += message.data[1:3]

Now it shows the DTC correctly but it only works for 1 DTC. Help with this would be appreciated.

barracuda-fsh, 2024-05-28:
I added another fix, so now it displays up to 2 DTC's correctly. If there are more than 2 DTC's stored in the ECU it will not display all of them.

barracuda-fsh, 2024-05-30:
I found a solution, it was not trivial. You can see it under changes. Now it works properly for 1 up to how many you want DTC's. Maybe this should still be under observation and if someone can compare to other programs if it works well? I tested under obdsim obd simulator.