barracuda-fsh/pyobd #20: Testing of DTC needed

barracuda-fsh, 2024-05-31:
DTC has been showing some errors, which should be hopefully fixed now. But I still need people to test it on their cars and comparing it with other programs, to be sure it works properly now. So I can be sure it is fixed. Please contribute to this if you can. I think under CAN protocol it is fixed, but other protocols need to be tested.

Please contribute.

barracuda-fsh, 2024-06-10:
Under CAN protocol I think it works now, under legacy protocol it shows only one dtc, even if there are many. If someone can fix this, I would be grateful.

Naveen-S06, 2024-06-20:
> DTC has been showing some errors, which should be hopefully fixed now. But I still need people to test it on their cars and comparing it with other programs, to be sure it works properly now. So I can be sure it is fixed. Please contribute to this if you can. I think under CAN protocol it is fixed, but other protocols need to be tested. Under obdsim app it works good now, but the app only uses CAN protocol.
> 
> Please contribute.

Can you give me the updated link So I can try it out?

barracuda-fsh, 2024-06-20:
Main branch has the updated code.

Naveen-S06, 2024-06-24:
pyobd only works on ELM327?

barracuda-fsh, 2024-06-24:
I don't know about other scan tools. Probably just elm327.

greendoescode, 2024-09-03:
Hey, what do ya need from me to help?

barracuda-fsh, 2024-09-03:
> Hey, what do ya need from me to help?

I need help from someone that has an older car and will trigger many DTC errors on it and then correct pyobd so that it can show multple DTC errors. I need a developer with an older car. Pyobd can't show multiple DTC errors on older cars. Are you a developer?

greendoescode, 2024-09-03:
> > Hey, what do ya need from me to help?
> 
> I need help from someone that has an older car and will trigger many DTC errors on it and then correct pyobd so that it can show multple DTC errors. I need a developer with an older car. Pyobd can't show multiple DTC errors on older cars. Are you a developer?

I have some knowledge with development. I have two cars I can try with, 2005 and 2009 respectfully. I can definitely have try, any specific DTC errors which are easy to produce?

barracuda-fsh, 2024-09-03:
Maybe engine intake air temperature meter, purge valve, o2 sensor.

greendoescode, 2024-09-03:
For sure, would the 2009 work? I'm not quite sure how my 2005 is doing :sweat_smile:

barracuda-fsh, 2024-09-03:
I think 2009 is too new, but I don't know. You will see under pyobd if it connects to can or any other protocol. If it's can, then it's too new, I think.

greendoescode, 2024-10-06:
Was able to get a DTC code for Idle Revs too high, but thats the only one so far

greendoescode, 2024-10-06:
2000 vw polo

barracuda-fsh, 2024-10-06:
yes, it only shows 1 dtc. it needs to be fixed.

barracuda-fsh, 2025-09-14:
I think you are close, yes. I forgot how it's decoded, maybe it's written somewhere on web. Or you can compare to the obd sim. It's a simple emulator that I trained the pyobd on, and it may be wrong.

barracuda-fsh, 2025-09-19:
Yes, 2001 is perfect. Thanks.

barracuda-fsh, 2025-09-28:
I will check your updated code, I'm just out of time now. To be continued....