barracuda-fsh/pyobd #30: NEED HELP of OBD Bluetooth adapter

ybsy915, 2024-07-13:
In your project, you recommended using Vgate icar pro 4.0, but the device manual we purchased only includes instructions for connecting to a phone. We need to connect to a computer via Bluetooth for development. How did you connect to the computer

barracuda-fsh, 2024-07-13:
Which OS are you using?

On Linux try using this or using bluetoothctl
https://wiki.debian.org/BluetoothUser

In bluetoothctl you just pair it and connect to it, maybe you have to use bind port command to get a port to connect to. I don't remember exactly how I did it, but this should help.

duncanhames, 2024-12-01:
Try this - https://wiki.archlinux.org/title/Bluetooth#Bluetooth_serial

pyobd worked nicely for me with an AUTOHIL AX2 Bluetooth adapter. Thanks @barracuda-fsh!