barracuda-fsh/pyobd #33: [ Feature request ]: add nix package.

Silverdev2482, 2024-08-08:
On nixos running unpackaged applications is annoying and hard, if not impossible. Adding a nix package would be very helpful.

barracuda-fsh, 2024-08-09:
Do the pyinstaller images not work? They are made available under every release.

Silverdev2482, 2024-09-06:
Sorry for forgetting about this issue, If you mean like pip install requirments.txt (that might be bad syntax) it won't work on nixos. I was able to make a shell.nix that worked though, maybe throw it in. although it isn't a proper package any you may need wifi to invoke nix-shell. 

```

[nix-shell:~/Programming/pyobd]$ cat shell.nix 
let
  pkgs = import <nixpkgs> {};
in pkgs.mkShell {
  packages = [
    (pkgs.python311.withPackages (python-pkgs: [
      python-pkgs.wxpython
      python-pkgs.pyserial
      python-pkgs.numpy
      python-pkgs.tornado
      python-pkgs.pint
      python-pkgs.pillow
      python-pkgs.six
    ]))
  ];
}
```

barracuda-fsh, 2024-09-07:
I meant the pyinstaller images for debian and ubuntu. Do any of those work? Here:
https://github.com/barracuda-fsh/pyobd/releases/tag/v1.19

Silverdev2482, 2024-09-09:
Oh they are just exectuables, no they don't work out of the box but they work just fine with steam-run, but a nix package with udev rules would be nice. Didn't test connected to an actual car but I'd assume they would work since the shell.nix worked.