barracuda-fsh/pyobd #35: feature request

devjohnschroeder, 2024-08-31:
Can we add the ability to show 8 graphs instead of 4 graphs on the "Graphs" tab?

Which line or lines of code can be modified to allow for this?

[image]

barracuda-fsh, 2024-09-01:
Yes we can, but it involves quiet a bit of work.

barracuda-fsh, 2024-09-02:
I made it. I need to check it a bit before I upload.

barracuda-fsh, 2024-09-02:
Done. 8 graphs are now under the latest code. Please test it and report back.

barracuda-fsh, 2024-09-02:
I hope your cpu is fast enough to handle 8 graphs.

greendoescode, 2024-09-03:
> I hope your cpu is fast enough to handle 8 graphs.

when you say fast enough, what are you running on?
My laptop might die if I try it lol

barracuda-fsh, 2024-09-03:
i7-4790k is what I'm running on and it says it takes 75% of the core to run it. I don't know. Just try and you'll see. Maybe it will run ok.

closed 2024-09-24 by barracuda-fsh