barracuda-fsh/pyobd #37: Doesn't work with J1939 Bluetooth Adapter

unixtech, 2024-09-17:
Connection is eventually working after trying many times but no data or sensors are working for Cummins engine.

[image]
[image]
[image]

Kindly direct me in proper direction, Thank you.

barracuda-fsh, 2024-09-17:
I'm not sure if I can help, but on bluetooth you need to set FAST to disabled and timeout to 5 or 10.

unixtech, 2024-09-17:
Still won't work, but Thank you for Answering. For reference, On Android Torque pro It shows all the sensors but Fault code won't show up. All I need to see is Fault codes.

barracuda-fsh, 2024-09-17:
Try "Car Scanner" app on android. It shows errors for free.

barracuda-fsh, 2024-09-17:
You could try editing obd_io.py line 77 to this - for manual protocol selection(1-9 and A are valid):
self.connection = obd.OBD(portstr=portnum, baudrate=baud, protocol="A", fast=FAST, timeout=truncate(float(SERTIMEOUT),1), check_voltage=False, start_low_power=False)

unixtech, 2024-09-18:
I already changed the protocol to "A", None of the Android app has support for SAE1939 29b 250KB protocol