barracuda-fsh/pyobd #38: Adding sensor data recording and replay feature

ktraister, 2024-09-30:
I'm happy to assist with this item I found on the TODO list in the README. I was thinking a button could be added to the "graphs" pages that would allow capture and replay, leveraging an XML file that could easily be imported into excel spreadsheets, or 
consumed by the replay functionality. 

Did you have thoughts/ideas/opinions on this? I wanted to ask before I started to write code and created a PR. Cheers!

barracuda-fsh, 2024-09-30:
To tell you the truth I figured I would never have enough time to do it. And also I'm not sure how I would do it. But the best idea that comes to mind is that under configure I would add(next to regular CONNECT) option RECORD and REPLAY.

RECORD would connect and only load sensors, all of them and put them in a JSON or something. When you disconnect, it would finish the JSON file with sensor data and exit. That's all.

Under REPLAY it would just load the JSON data from file and I would only allow sensors and graphs that would be taken from that JSON file. I would also add some delay every time it loads a frame from json, so it wouldn't go crazy.

I think this is the best approach. Can you do this?

ktraister, 2024-09-30:
Yeah, that's something I can do. I want this feature myself, and I have the python dev experience to make it happen. I may need some time to get it done, but I'll post an update on this thread in the next few weeks.

As far as the approach you've defined, that works for me. I'm thinking a 1 second delay between writes should be sufficient to capture detail and also not overwhelm the host's disk. I'm less particular about the GUI as long as it all works

barracuda-fsh, 2024-10-02:
If you need any help, you're free to write to me on my e-mail. But mostly it will be on you. I hope you can do it.

ktraister, 2024-10-02:
I appreciate the offer, but I don't think I'll need help. I'm about halfway done with the GUI functionality and it's been smooth so far. Expect a PR from me in a few weeks!

barracuda-fsh, 2024-10-02:
Ok, good.

jgonzales20, 2024-10-29:
> I appreciate the offer, but I don't think I'll need help. I'm about halfway done with the GUI functionality and it's been smooth so far. Expect a PR from me in a few weeks!

Please add support for export in .svg format for the graph or pdf

barracuda-fsh, 2024-12-13:
> I appreciate the offer, but I don't think I'll need help. I'm about halfway done with the GUI functionality and it's been smooth so far. Expect a PR from me in a few weeks!

Hi! Did you manage to do it?

sszxc, 2026-01-18:
Thanks for the repo first. I'm also looking for some better and more modern ways of recording data.

I tried hooking up Rerun as a visualization backend, which easily adds data replay/timeline features, and it's super useful for analyzing.

Rerun is probably too heavy as a dependency, but if anyone wants to play with it, feel free to check out my fork: [https://github.com/sszxc/pyobd-rerun](https://github.com/sszxc/pyobd-rerun)

https://github.com/user-attachments/assets/866877e1-9753-45dd-8399-618a3995f3d4