barracuda-fsh/pyobd #40: Issues with Connecting the Vgate iCar pro ble(Bluetooth 4.0)

DominikBabic06, 2024-10-19:
I am using a 2024 M3 MacBook Pro with macOS Sonoma 14.5. I have an iCar OBD-II device that works flawlessly with my iPhone. I can easily discover and connect to it using the LightBlue app on my iPhone.

However, when trying to connect the same device to my MacBook, it does not show up. Even when using a Bluetooth scanner like LightBlue on my MacBook, the iCar device remains undiscovered. Strangely, I can find and connect to other Bluetooth devices on my Mac without any issue, but the iCar OBD-II device is not one of them.

Any assistance in resolving this would be greatly appreciated!

barracuda-fsh, 2024-10-20:
It's weird that it doesn't work with the Macbook Pro. One thought though is that after you plug in the icarpro and turn the ignition on, that it turns itself off or goes to sleep if you're not fast enough.

DominikBabic06, 2024-10-21:
It does seem a bit unusual, but how did you manage to connect the device to your MacBook? On both my iPhone and Mac, I couldn't locate the device using the default Bluetooth manager. I was only able to find it on my iPhone using the LightBlue app and with a car scanner app.

barracuda-fsh, 2024-10-21:
I didn't connect it to my macbook, I connected it to a Lenovo Thinkpad.

barracuda-fsh, 2024-10-21:
Did you manage to pair it at least?

DominikBabic06, 2024-10-21:
I haven't had the chance to try again since my initial post about the issue. The device didn't show up on my MacBook at all the last time I checked. I might try connecting it to an old Windows device to see if it's an operating system issue.