barracuda-fsh/pyobd #42: Installation fails with Python 3.12

jhunnius, 2024-11-06:
Installation fails with Python 3.12 due to a missing distutils, which was deprecated.

joeldjro, 2024-11-17:
I don't know if you are getting the same error at the end of the installation where pyinstaller-debug is owning a file created during te pkbuild. I had to remove pyinstaller-debug before installing pyobd.

rquarry, 2025-01-09:
I can confirm this issue occurs when `pip` attempts to install numpy via `requirements.txt`. The issue and resolution is detailed on [this stackoverflow post](https://stackoverflow.com/questions/77233855/why-did-i-get-an-error-modulenotfounderror-no-module-named-distutils). 

I was able to get past this error with `pip install setuptools`

closed 2025-05-13 by barracuda-fsh