barracuda-fsh/pyobd #43: Help Please

deanfourie1, 2025-01-01:
Hi there,

I have been trying now for months to get this working, I have a Raspberry PI Zero W 2 and a 2.5inch eInk display I plan to display the ODB data on,

I am having real trouble getting this to work. I have a few questions,

1. Is this project still in active development? Will the current steps outlined still work to this date?
2. Should I follow the installation instructions on this repo or from here? https://python-obd.readthedocs.io/en/latest/
3. I am using Bluetooth ELM327. I know this is not recommended however I know the Bluetooth adapter works using Real Dash on android, but I can also get a USB one that is recommended.
4. Is a desktop environment required? I am running all lite versions of raspian, when launching I get an error related to startx failing to start.

If anyone can please help me or better yet even has a image they can share with me with a working install of PyOBD. I have been trying for so long to get PyOBD to work just to get my car temperature displayed.

Thanks

barracuda-fsh, 2025-01-01:
Hi!

This program was meant for desktop computers/laptops. I never tested it on raspberry pi. Probably you will need to modify it to work on rpi and that is not easy. 
The program is in semi active development. it should work on python 3.8 to 3.11, I am not sure about 3.12. Last update was in september 2024. If you ask me then this is pretty current. On bluetooth it's harder to connect but it does work. You need to pair your adapter and then connect to it, you will need to use command line utilities for that and google to find out what exactly you need to do.
Yes you pretty much need a desktop environment for pyobd.

NA18784, 2025-01-01:
>  I have a Raspberry Pi 4 with the 7” touch screen and have been able to get pyobd program to kind of work. The program starts up, but the screen resolution is off (too big for the display). I can change it and it then displays correctly, but doesn’t stay with the resolution once program has been restarted.  I have not been able to connect to a vehicle with it though (using a elm327 usb to obd2). Been put aside until I get a bit more time to tinker. Also watching these notifications to see if there had been updates on it.

[image1.jpeg]

deanfourie1, 2025-01-01:
> I have a Raspberry Pi 4 with the 7” touch screen and have been able to get pyobd program to kind of work. The program starts up, but the screen resolution is off (too big for the display). I can change it and it then displays correctly, but doesn’t stay with the resolution once program has been restarted.  I have not been able to connect to a vehicle with it though (using a elm327 usb to obd2). Been put aside until I get a bit more time to tinker. Also watching these notifications to see if there had been updates on it.
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> Hi!
> 
> 
> 
> This program was meant for desktop computers/laptops. I never tested it on raspberry pi. Probably you will need to modify it to work on rpi and that is not easy.
> 
> The program is in semi active development. it should work on python 3.8 to 3.11, I am not sure about 3.12. Last update was in september 2024. If you ask me then this is pretty current. On bluetooth it's harder to connect but it does work. You need to pair your adapter and then connect to it, you will need to use command line utilities for that and google to find out what exactly you need to do.
> 
> Yes you pretty much need a desktop environment for pyobd.

Hi are you able to share a image with me? I can work on getting the Bluetooth working? I am also thinking I may just buy a serial usb obd anyway as there is a bit of latency with the Bluetooth modules to be honest.

If you could share an image with me that would help a lot, been struggling with this.

Thanks

NA18784, 2025-01-01:
I’ll turn it on maybe tonight and see if can make an image for ya.

deanfourie1, 2025-01-02:
@NA18784 that would be so awesome if you could. Thanks heaps

AWildNoodle, 2025-01-02:
> I have a Raspberry Pi 4 with the 7” touch screen and have been able to get pyobd program to kind of work. The program starts up, but the screen resolution is off (too big for the display). I can [...]

I ran into the same problem with my raspberry pi setup and a 7in IPS touch display. Was there something in the python scripts you modified for the resolution or did you do it through the rpi gui/settings?

also interesting about the USB elm327 because i cant get the bluetooth to work on mine but i can with the usb lol. 
this is the one i bought --> https://www.amazon.com/gp/product/B07MQ8GHG3/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1 

could take a look at the groups your user is added to as well maybe that is issue?

NA18784, 2025-01-02:
There was a section in the script to change resolution, but for some reason it reverts back to the larger default. I’ll have to try and dig back in my emails to find which line it was.

barracuda-fsh, 2025-01-02:
It's probably this. pyobd.py, line 2639:
self.frame.SetSize((1024, 920))