barracuda-fsh/pyobd #51: Can't connect Dacia Duster 1.5dci K9k engine with 500k baud over Bluetooth

zaszlo, 2025-06-18:
The connection succeeds with 500k baud with Car Scan android app. 
Can't connect Dacia Duster 1.5dci K9k engine with 500k baud over Bluetooth. 
Tried with automatic Baud rate, still can't connect.
Manufactured 06 2011, Engine K9K896, VIN: UU1HSDAC645573318

Any help what could I try to get it connected or is not supported?

barracuda-fsh, 2025-06-19:
Hi! You have to use terminal to connect to the bluetooth adapter and it must stay connected to it. And then you must enter another terminal command for it to connect to serial port of the bluetooth adapter. You can read this online, use google. I don't remember which commands you need, but connecting to the adapter with gui will disconnect it after so you must do it with terminal. I presume you are using Linux? Try to google it, you may need half an hour to figure out exactly what to do, but it does work this way.

Put this in google search and AI will tell you everything:  "bluetooth terminal connect linux obd serial port"

zaszlo, 2025-06-19:
Using Windows, and in the Bluetooth settings I see COM5 with Outgoing OBDII 'JL_SPP' and COM6 with Incoming OBDII.
Tried manually setting COM6 but no luck.
Wierd thing is that after connecting successfully to the OBDII and OBDBLE devices that appear by searching for bt devices, the description is always 'Not connected' and has no 'Connect' button as the audio bt devices do for example. 
FORScan displays some debug info when trying to connect:
- Established connection to COM5, adapter identified ELM327 v2.1
- Unable to connect.
In the android app I wasn't able to connect either until I selected the right model for the car, but in the desktop apps I didn't met any settings for setting the car profile.

barracuda-fsh, 2025-06-20:
I honestly don't know. Maybe you need a better adapter like Vgate iCar Pro, but I can't make any guarantee.